lachlanA/eagle-to-kicad #5: KiCad PCB import bugs

lachlanA, 2015-11-04:
There are a number of bug's on KiCad PCB import,  which I hope to some patch's accepted.
will give status report in a few weeks..   As RC2 is not far away. Of Kicad

timofonic, 2015-11-05:
Thanks for all!

Are you considering to help at implementing full Eagle import-export in KiCad itself too? This way, interoperability circle would be complete ;)

lachlanA, 2015-11-11:
The main reason I used the ULP's, were a number of existing ULP's out
there, which did the conversion.  But they had problems.
So I picked the best one, and keep fixing/adding extra stuff in.  ULP's had
the advanage, it could convert old version of Eagle files to Kicad,  as
all file format's less  than 6.xx were binary, and needed to be reverse
Eng, to get them to work with KiCad.
I think there was a Python only script,  which did not depend on Eagle,
but I had a look, but could not get it to work,  and endless dependence's

I think one of the guy's at Kicad, ( I dont know who)  is doing a full
eagle sch import,  after the 4.0 release of KiCad.  Not sure of the status,
thou you can always.  Ask the mailing list.

At the moment's I'm trying to fix up the KiCad side of pcbnew import,  as
it still has problems.   And I hope to get those fix's in before the 4.0
release.
But alas  I'm sic at the moment.   So not much doing utill I get better.

So in short answer, is no, I wont be doing a Kicad only version, as it's
already being done as far as I know.  And have been told not too, by the
developers at KiCad.

Lachlan

closed 2016-11-07 by lachlanA