lachlanA/eagle-to-kicad #6: PCBNew bug on Mac OSX

fcarlier, 2015-12-28:
With my pcbnew 4.0.0-rc2-1 on Mac OSX : 
I have this error message : 
Error loading board.
IO_ERROR: Unknown file type
from /Users/jenkins/remoteroot/workspace/KiCadBuildMac4/kicad/pcbnew/legacy_plugin.cpp : checkVersion() : line 570

fcarlier, 2015-12-28:
idem with pcbnew 4.0.1 ;)

lachlanA, 2015-12-28:
Can you send me the file ?
My email is laudas1@gmail.com

So I can check,   I only have linux,  but will give it a try,

Lachlan.

lachlanA, 2015-12-29:
Hi fcarlier

Just double check, that in pcbnew  when opening the eagle file that you
select the  "eagle 6.x XML PCB files".brd   format.
It's confusing, I know,   that KiCad used the  "*.brd"  file type for  it's old board's which is the same
as Eagle file type,  plus adding to the confusing,   kicad only understands   eagle file's with 6.xx or above(XML format).
Where as my script understands Eagle's old binary format as well as the  new format.

So that's why the script put's the Eagle file's  into 3 target directory's.
    TARGET_DIRECTORY
     eagle_original_files/   .......................   This the copy of the Original eagle files, you picked,  so they will be exactly the same.
                                                               maybe old format.
     kicad_files->modified_eagle_files/  ....   This contains the Eagle file's converted to 6.xx format (XML format) and the molded, The script
                                                              make's those file's, to make life a little easy for finding error's and get around some of the limitations
                                                              of KiCad.   They will be the newer Eagle format,  and you need to select the Eagle PCB file from here.
                                                              not from the "eagle_original_files/" directory.   Note you have to select the  "6.xx or above(XML format)"
                                                              When opening the file in pcbnew,  from  kicad_files->modified_eagle_files/
 
I'ts a pity I can't scrip this, but  pcbnew dose not allow me to load a Eagle PCB file, from the command line!

I should fix the video,   to make it easier to understand, but I'm working on the KiCad side of thing's at the moment.
As it still dose some nasty thing's like,   re-sizing tracks with out warning you!,  etc..

Lachlan

fcarlier, 2015-12-29:
> HiFlorent
> On Mon, Dec 28, 2015 at 2:40 AM, Florent Carlier
> <florent.carlier@univ-lemans.fr <mailto:florent.carlier@univ-lemans.fr>>
>     :) No Problemo :)
>     Just for information : it's identical this description bug :
>     Good night
> Thanks Florent,
> It's late here in Australia so will have a look at it in morning :)
> On Mon, Dec 28, 2015 at 1:55 AM, Florent Carlier
> <florent.carlier@univ-lemans.fr
>     <mailto:florent.carlier@univ-lemans.fr
>     Hello,
>     I am testing with your example file, it's identical case.
>     I check kicad github :
>     ( LEGACY_PLUGIN::checkVersion() )
>     if i add 'PCBNEW-BOARD Version 1' at begin file of modified-eagle/*.brd
>     it's ok but i have a same problem with 'LEGACY_PLUGIN::loadAllSections'
>     My version is Max OSX 10.11.2 with eagle 6.6.0 and Kicad 4.0.1
>     Schematic is OK, good import; Library is OK, No PCB :)
>     Florent
> My email is laudas1@gmail.com <mailto:laudas1@gmail.com>
>     <mailto:laudas1@gmail.com <mailto:laudas1@gmail.com>>
>     <mailto:laudas1@gmail.com <mailto:laudas1@gmail.com>>>

Hello,

I'm using well the *.brd of modified_eagle_file ;)

I think that it's a bug of Kicad on MacOSX. kicad can't maybe detected
the eagle file ! and call directly legacy_plugin.cpp !!

If you look the source file, you can see that it need the 'PCBNEW-BOARD
Version 1' parameter and more.

I go to wait this information on this topic
https://bugs.launchpad.net/kicad/+bug/1526046

THX

lachlanA, 2015-12-29:
Ok no problem's    sorry I don't have a Mac to test stuff out on !
I could be of more help..  

Lachlan.

BrainyBrian, 2016-11-06:
..?

closed 2016-11-07 by lachlanA

BrainyBrian, 2016-11-07:
So what happened to this?

lachlanA, 2016-11-07:
Don't no BrainyBrian,  did not get any files from  fcarlier,  and I don't have access to a mac for testing.
so have no more info to give you.

Lachlan

cyoung, 2016-11-08:
I'm getting this error on a .brd file directly saved from Eagle 7.5.0.

lachlanA, 2016-11-08:
Hi cyoung,  are you running the script on a Mac ?
is it only 7.5.0 ?    and is it the modified converted eagle .brd file build  by the script,  or the original file ?
Oh one last one can you sent me the file sch/pcb files  ?

Lachlan

cyoung, 2016-11-08:
@lachlanA - yes, on macOS. I've found this bug: https://bugs.launchpad.net/kicad/+bug/1551628

It's just a KiCad interface bug for macOS. I got around it by sending the file to a Windows machine and doing the conversion there, then sending it back. It has been fixed in KiCad but I guess the fix hasn't made it to release yet.

lachlanA, 2016-11-09:
Thanks for the info cyoung !
I think there working on a new stable version  to come out in a few weeks...
with luck it may be fixed then.

Lachlan

coddingtonbear, 2016-12-16:
For what it's worth, you can work around this problem by installing the latest Kicad nightly development build (on the bottom of their [normal download page](http://kicad-pcb.org/download/osx/)), and then using the "Import Non Kicad Board File" menu item.

lachlanA, 2016-12-16:
Thanks for the info Coddingtobear, 
I don't have a mac to test,  but I'm sure it will help some one else with a mac.

Lachlan