lachlanA/eagle-to-kicad #8: Converting large library from eagle to kicad

Squantor, 2016-05-06:
I am getting the following error:
eagle-to-kicad/eagle-lbr2kicad-1.0.ulp(2116)

Reference to uninitialized object variable.

The library is pretty big but I am not at liberty to share it publicly. Alas my eagle script skills are not sufficient to tackle this problem by myself, can I send you the library?

Squantor, 2016-05-06:
Well I am currently converting all my libraries, and this open source library shows the same problem.
[SquantorIC.lbr.zip](https://github.com/lachlanA/eagle-to-kicad/files/251792/SquantorIC.lbr.zip)

lachlanA, 2016-05-06:
Hi Bart,   I will check into it...  see if I can figure it out.

Lachlan

lachlanA, 2016-05-06:
What version of eagle are you using..  and what's it running on ?

Lachlan

lachlanA, 2016-05-06:
Hi Bart,  I have fix the problem I think,
I was assuming device's with out package,   would only have 1 or less pin's
but that assumption was wrong.   Try  the new version..   and see if you have any better luck.
Oh and thanks for the bug report :)

lachlan

Squantor, 2016-05-06:
Hi Lachlan,

Um sorry for the noob question but is that version checked in? I cant see any feature branches either. Thanks again for the quick fix btw!

Bart

lachlanA, 2016-05-06:
should be,    my github shows the changes applied 6 hours ago.
you should see in the comment's,  "fix bug where, Device has no package and more the 1 pin, update version" 
And should show if you diff you local version and the one on github.

Lachlan.

Squantor, 2016-05-06:
Hello Lachlan,

Found the reason why it did not show up, I had to pull https://github.com/lachlanA/eagle-to-kicad-libs
Tested it and worked like charm!

Thanks again.

Bart

closed 2016-05-06 by Squantor