lachlanA/eagle-to-kicad #9: eagle6xx-sch-to-kicad-sch.ulp asks wether to apply net list label fix, but ignores choice

pgroe, 2016-06-09:
Line 3228:
`dlgCheckBox("", enableNetListLableFix ); dlgLabel("<nobr><b>Enable Netlist Label Fix</b></nobr>"); dlgSpacing(20); dlgStretch(0);`

is the only real occurance of the variable `enableNetListLableFix`.

It is declared in Line 377: `int enableNetListLableFix = 1;` but that's all.

No evaluation of its value. Just mentioning.

I stumbled across this, as I like to fix the labes by myself.

pgroe, 2016-06-09:
Suggesting patch:
`
--- "eagle6xx-sch-to-kicad-sch-org.ulp"     2016-06-09 19:23:45.046044800 +0200
+++ eagle6xx-sch-to-kicad-sch.ulp       2016-06-09 19:22:52.083951800 +0200
@@ -3001,7 +3001,9 @@
                   write_kicad_wire_segments( SH );
                   write_kicad_bus_segments( SH );
                   write_kicad_dotted_segments( SH );
-                  write_kicad_netNames( SH, current_sheet );
+                  if (enableNetListLableFix) {
+                      write_kicad_netNames( SH, current_sheet );
+                  }
                   printf("$EndSCHEMATC\n");
                 }
               }
`

pgroe, 2016-06-09:
Suggesting patch:

[eagle6xx-sch-to-kicad-sch.txt](https://github.com/lachlanA/eagle-to-kicad/files/307548/eagle6xx-sch-to-kicad-sch.txt)

lachlanA, 2016-06-10:
Thanks pgroe,  sorry for the slow replie,
I have been flying from Europe to Australia for last day or so,   so I'm a  bit whacked.
Will look at your input tomorrow..    when my  brain works out what's up and down, and which day it is.. !
Thanks again..

Lachlan

lachlanA, 2016-06-11:
I have committed you patch and updated the version and build date to reflect your fix.
thanks for the help,  is nice to have feedback and fix's !
The script's are a really nasty hack,  Nothing I'm proud of,  as you can see by the mess of code.
but Given the limits' of Eagle to KiCad it works most of the time.
Thanks again.
Lachlan

closed 2016-06-11 by lachlanA

pgroe, 2016-06-13:
No problem ;) Didn't want to hustle you.
I'm happy I could contribute, but it's only very simple 2 lines of code, whereas you contributes thousands of lines. So thanks to you!

lachlanA, 2016-06-15:
Any bug fix is good !
No matter how small,    the coded is such a mess,   and I don't get any
feed back,   just stared,  which is nice
but having a email  saying where the problems are..  and works..  is a
1000% better !
Thanks again for the help.

Lachlan.