lachlanA/eagle-to-kicad #10: Error loading board.

phillip-toone, 2016-08-24:
I tried going through your example video and encounter the following problems.

At 10:08 when I open the kicad project I note that the following files are not listed as part of the project:
  RGB LED 1.4.lib
  RGB LED 1.4.net
  RGB LED 1.4-cache.lib

At 11:52 when I run CvPcb to associate the components and footprints I get the following error message:
"No PCB footprint libraries are listed in the current project file."

At 12:26 when I open pcbnew I get the following error message:
"Error loading board.
PARSE_ERROR: Expecting 'number' in input/source "/home/user/RGB LED 1.4.kicad_pcb", line 67, offset 23
from /build/buildd/kicad-0.20131208+bzr4024/common/dsnlexer.cpp : Expecting() : line 285"

phillip-toone, 2016-08-24:
I think I've discovered that my problem was that I was using an older version of KiCad (~2013). I've updated to the latest version of KiCad and now everything seem to work flawlessly.

lachlanA, 2016-08-25:
Hi  Phillip,   thanks for the info,   They are quite a few bugs out
there,   I have found,   so pays to try a few version's of KiCad or even
Eagle !
I'm Glad the script is working   😂

Lachlan

closed 2016-08-25 by lachlanA