lachlanA/eagle-to-kicad #14: renumber ulp; Unknown identifier 'EAGLE_HOME'

dbrown2k, 2017-01-20:
Even with a fresh install of 6.3.0 i'm getting the following error; 

Unkown identifier 'EAGLE_HOME' 

pointing to the following lines in the ULP;

string outputPath = EAGLE_HOME + '/';
string myULP_HOME = EAGLE_HOME + '/';

when trying to run renumber-sheet.ulp. I've tested against the RGBled example with the same result. and tried adding EAGLE_HOME to the environmental variables. Any ideas?

machine; win7 64bit

lachlanA, 2017-01-21:
I tested it on 6.6 under Linux,  and old 32bit windows XP,  so can you try download version 6.6.0 of Eagle and see if you still get the same Error ? on win7 

Lachlan

dbrown2k, 2017-01-21:
That runs on 6.6 on win 7 64bit, thanks 

And for anyone looking for older install files: ftp://ftp.cadsoft.de/eagle/program/

closed 2017-01-21 by dbrown2k