lachlanA/eagle-to-kicad #15: Absolute path names in converted Kicad files

BrianHoldsworth, 2017-01-21:
I attempted to run the ULP scripts on a Mac and created all the Kicad files. I moved the Kicad files over to a Linux box to continue there with Kicad. Naturally, moving all the Kicad files changed their absolute path filenames on the Linux machine compared to where they were originally saved on the Mac.

The first indication of the problem was when running CvPCB. I immediately got an error about not being able to locate the .mod file.

Looking through the Kicad files, it is not clear where this absolute path has been stored. I assume that files created by the ULP scripts ought to use relative path filenames to avoid this problem when relocating the files. I think this will be especially common problem since Eagle is more likely to be run on Windows and Kicad more likely on Linux.

lachlanA, 2017-01-21:
The only testing I did was on Linux,  and some small files with the old windows XP
I know the Mac has be a major pain for the KiCad developers, and have read of them pulling there hair out
to find problems,   the Project file I build,  is just what I figured from from looking at the Linux stuff,
if you have a working fresh project file from a Mac, perhaps I can compare the diffidence's 
and maybe make the fix's.

Lachlan

BrianHoldsworth, 2017-01-21:
Hi @lachlanA . I attached a ZIP file that has all the output from running the ULP scripts on Mac. These are files that I FTPd to my Linux box, then got stuck on the Kicad side.

[Archive.zip](https://github.com/lachlanA/eagle-to-kicad/files/721494/Archive.zip)

lachlanA, 2017-01-22:
Thanks Brian,
Just got up,  nice sunny Sunday here in OZ !
will will look at it latter today,  thanks :)

Lachlan

lachlanA, 2017-01-23:
Hi Brian,
script make's a old kicad lib format,  which needs a file called "fp-lib-table"
it holds the path to the *.mod file in that,  try changing the path there.
When  KiCad CvPCB open's it,  it  will warn you about the old format and ask to save in the new one
If I remember correctly.
If I get any one else with this problem, I will update the doc's,   of if you like you can edit  docs
file,  and make a note about moving the files.  And I will commit your changes.

Lachlan

BrianHoldsworth, 2017-01-24:
Hi @lachlanA. I confirmed that was the problem. After modifying the absolute path in "fp-lib-table", AND also moving my Kicad files to my $HOME directory (due to Kicad file permissions issue), I was able to complete the Kicad import without any unexpected problems. Nice work!

There may be 2 things worth changing in the docs for Linux Kicad users:
1. Place the Kicad files created by the Eagle export process in a directory in your $HOME path.
2. Manually edit "fp-lib-table" to have absolute path to your relocated .mod file.

But, being new to Kicad, I'm not sure others will bump into this. For reference, I am using Kicad on latest Ubuntu x64, and I installed Kicad the "dumb way", just using the Ubuntu Desktop software manager.

lachlanA, 2017-01-24:
Thanks Brian,
I think the problem only shows up if you move the file's before you complete all the steps
as the new version of KiCad dose not use the fp-lib-table,  if I remember correctly, 
So the problem would only show up in that case. 
Perhaps I should check for Write access,   before continuing the script's
That would cover your case and others in which they don't write permissions. 
Being limited to read only permission for the source eagle file's and needing  to copy to a $HOME directory
with out doing this step manual, is a good idea,  I will give it some  thinking.

Thanks once again for the feed back,  I love hearing from the long suffering  user's ..  be it good for bad ;)

Lachlan