lachlanA/eagle-to-kicad #16: Unknown part in schematics

Squonk42, 2017-01-25:
Hi Lachlan,

Thank you for your great work!

I am in the process of moving away from EagleCAD due to the new subscription-only business model, and I use your latest scripts to convert my existing projects from EagleCAD 7.5 ti Kicad 4.0.5 on Linux.

I face 2 issues (do I have to open separate tickets in Github?):

1. all my passive resistors and capacitors are replaced in the schematics by squares with question marks for pins
2. all my power supplies except GND are doing the same

I kind of figured out what is going on for issue 1), my EagleCad parts are using variants (e.g. CAP_0402 or RES_0402), and this is the name that is assigned to them by the conversion tools, whereas the converted part library name does not have the variant as suffix (e.g. CAP_ and RES_). By removing the extra variant in the schematic part name, I get the right part displayed correctly. So, it looks like to me the variants are not handled correctly by the conversion tools, are they?

As for issue 2), I don't know how to fix them, since the missing power supplies are not in the converted library. I thought it was because they were prefixed with a "+" sign (like +3.3V), but this is not the case, as I am also missing a "PE" power supply. Do you have a hint how I could correct this?

Best regards,
Michel

lachlanA, 2017-01-25:
Hi Michel
Can you send me the files,  ?   so I can look at them,
I had to do a lot of name hacking,  on part name's,   as Eagle permits lot's of charter's which
KiCad dose not,    so it's not the best,  but maybe it's  one I mist, which should be converted to a "_"
in the lib and not being converted in the net list, as the export has lib export and sch export
and the replacement charter check,   my not be in sync.   Whats why your not seeing it.

Lachlan

Squonk42, 2017-01-25:
Hi Lachlan,

Here are my original files in EagleCAD 6.1 format, that I open with EagleCAD 7.5 Freeware:
https://github.com/Squonk42/USBug

And the corresponding Github page is here (kind of slow...):
http://squonk42.github.io/USBug/

thunder7211, 2017-01-26:
I am having the same problem as mentioned by Squonk42

Squonk42, 2017-01-26:
Regarding the second issue, the "PE" power supply is in fact not missing.

Thus, only the power supplies whose name are prefixed by a "+" sign are not converted in the library, but referenced in the schematics.

Not exactly the same as Issue 1, but still a name folding problem. Can you explain how this folding is done and what are the rules?

thunder7211, 2017-01-26:
After taking a brief look at the .sch files, I found that if you change "DISCRETE_C-0402" to simply "C", the mystery box is replaced with the correct component symbol. Looks like there may be some naming issues during the conversion.  I don't know if this is due to using a newer version of Eagle. I discovered this was the correct format by opening a .sch file that I had made in KiCAD.

Squonk42, 2017-01-26:
Yes, I found too that if I edit the Kicad .sch file and remove the EagleCAD "variant" from the part name, I get the right part rotated correctly... And if I understand correctly, changing to simple "C" like you did, the capacitors will be replaced by default Kicad capacitor, unfortunately not rotated correctly this time.

lachlanA, 2017-01-26:
Yes the Eagle variant,  has given me lot's of problems,  as Ealge software defines a format for it,
but dose not check it, in the Early version's of the Eagle software, so you could get away with
murder,    But the latter version would check it,  but would accept the old mistake's from the old files.
Thou I note the 7.xx would do some extra checks and complain too, in some case's.

I fixed a few case's, and yet again this looks like a replay of the "variant" mess !
so I will try to add  a work around,   and not mess something else up at the same time.
As the code is a house of cards at the moment...   Sad to say.

Lachlan

thunder7211, 2017-01-26:
Ah yes, Squonk42, you are right! Changing to "C" replaces it with the default KiCAD capacitor and rotates it incorrectly.