lachlanA/eagle-to-kicad #20: Unknown identifier 'EAGLE_HOME'

silverdr, 2017-02-18:
When trying to run the `renumber-sheet.ulp` I only get the following error:

```
/Users/silverdr/sources/eagle-to-kicad/renumber-sheet.ulp(97):

Unknown identifier 'EAGLE_HOME'
```
I guess I may need to set it somewhere but it would be nice to take a case where this is not set also into account.

[image]

lachlanA, 2017-02-18:
Hi, silverdr,  I don't have a Mac,   so I can only have a SWAG (Scientific wild-ass guess)
as to what is going wrong,    
https://www.element14.com/community/thread/21425/l/eagle-version-64-released?displayFullThread=true
Looks like it may be fixed in Version 6.4 Of eagle,  if you can,  try version 6.4 or latter, (but not 8)
see if that fix's the problems.
And let me how you go.

Lachlan

silverdr, 2017-02-18:
I don't think it is Mac specific. I managed to get further by removing the offending constants. They were also in another script. Then I encountered a requirement for a version 6.3. I worked that around too but then I eventually got to a point when clearly a newer version is required. So it seems that to use the scripts one needs at least 6.4, despite one of the requirements saying 6.3. Since Cadsoft no longer provides the 6.4 downloads I need to check if I can get it somewhere from..

lachlanA, 2017-02-18:
Thanks for the hard work silverdr,    As I said I don't have a Mac,   so was unable to test the scripts
there, and until you started giving me bug reports I had no idea how well it worked,  or which version
of eagle was workable with Mac,     if you could give me,  some more info,   as to what you find maybe
I can fix them and send you the test version,  and update the doc's to reflect your finding/fix's

Thanks again.

Lachlan.

silverdr, 2017-02-19:
Have you tried the scripts on your OS but with EAGLE version 6.1? As I wrote, I don't think these problems to be OS specific. If you have a chance to try it on 6.1 it would probably help as I expect them to show up also on Linux and Windows.

lachlanA, 2017-02-19:
Hi Silverdr,    I'm only using the very latest version of Eagle 6.6  if I remember correctly
I will see if I can can download the 6.1 version,    and let you know how I go.

Lachlan

lachlanA, 2017-02-19:
Hi Sliverdr,   I tryed finding 6.1 for Linux but no luck :(
they have removed it from the Eagle website,  looks like only 8 is available from what I can see.
I also notice there are quite a few bug reports filed for 6.1,     so dose not look like it's a release you can trust.
But if you can find it,   please give me the link so I can download it and try.

Lachlan.

silverdr, 2017-02-19:
I have 6.1 for Linux somewhere. I used it along with the osx version. I shall find and put it up for you.

lachlanA, 2017-02-19:
Supper :)

lachlanA, 2017-02-24:
Hi, silverdr   I have removed the need for EAGLE_HOME   which was added in 6.4 version of eagle,
so give it a try, and see if it runs under Osx

Lachlan

silverdr, 2017-02-24:
I apologise for the delay.. I got hit by high fever and other flu symptoms since Monday. It's the first day now I am more or less OK. Removing the EAGLE_HOME requirement is what I did first myself but that was not enough as I mentioned above. I checked the files I promised and I think this I found my old Installation file for Linux. I'll try to send a message directly to you rather than post the file here ;-)