lachlanA/eagle-to-kicad #21: assert "IsValid() && dt.IsValid()" failed in operator!=(): invalid wxDateTime

PeteLawler, 2017-02-21:
I'm trying to convert an Eagle file ([Ray's Hobby Opensprinkler Beagle v1.1](https://github.com/rayshobby/opensprinkler-beagle/tree/master/hardware)) from Eagle to KiCAD.

During the conversion, at the "Assign Component Footprint" stage, I receive multiple versions of this same error. I'm unsure whether it is the script or something else, however I feel the script should maybe handle the issue. I would certainly feel more confident about the output if these errors didn't happen.

Using Eagle 7.7.0 and KiCAD 4.0.5+dsfg1-4 wx Widgets 3.0.2 Unicode Boost 1.62.0 on Debian Stretch amd64 x86_64 64 bit
 

ASSERT INFO:
/usr/include/wx-3.0/wx/datetime.h(876): assert "IsValid() && dt.IsValid()" failed in operator!=(): invalid wxDateTime

BACKTRACE:
[1] wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const
[2] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[3] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[4] wxEvtHandler::TryHereOnly(wxEvent&)
[5] wxEvtHandler::ProcessEventLocally(wxEvent&)
[6] wxEvtHandler::ProcessEvent(wxEvent&)
[7] wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const
[8] wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&)
[9] wxEventHashTable::HandleEvent(wxEvent&, wxEvtHandler*)
[10] wxEvtHandler::TryHereOnly(wxEvent&)
[11] wxEvtHandler::DoTryChain(wxEvent&)
[12] wxEvtHandler::ProcessEvent(wxEvent&)
[13] wxWindowBase::TryAfter(wxEvent&)
[14] wxEvtHandler::SafelyProcessEvent(wxEvent&)
[15] wxMenuBase::SendEvent(int, int)
[16] g_closure_invoke
[17] g_signal_emit_valist
[18] g_signal_emit
[19] gtk_widget_activate
[20] gtk_menu_shell_activate_item
[21] g_closure_invoke
[22] g_signal_emit_valist
[23] g_signal_emit
[24] gtk_propagate_event
[25] gtk_main_do_event
[26] g_main_context_dispatch
[27] g_main_loop_run
[28] gtk_main
[29] wxGUIEventLoop::DoRun()
[30] wxEventLoopBase::Run()
[31] wxAppConsoleBase::MainLoop()
[32] wxEntry(int&, wchar_t**)
[33] __libc_start_main
[34] _start

lachlanA, 2017-02-21:
Hi PeteLawler,   I will look into tomorrow,     let you know what I find.
Oh thanks for the corrections.. to the Doc's :)

Lachlan

lachlanA, 2017-02-22:
Hi PeterLawler,   The lib's embedded in this design, have a number of problems
MAC97 symbol="TRIAC"  symbol requires,  the device has 4 pins,  but the PCB device lib has a TO92(which is 3),  and a SOT223 has 4,  so the conversion script spit's the dummy over this,  Also there are 2 inductors one with device number L1  and L1'    I don't think KiCad like's  '   in the device reference numbers, I do not check for '  in my script,  but I'm double checking to see if I need to add,   all so there is $1_6_$ request  during the conversion,  which I have not looked at yet.
This show's up the problems of Eagle lax lib checking code very well !
OLIMEX LTD,  libs were very bad,   as they even managed to put in SCH layers into the LIBS parts
which should be impossible !   But I expect it's Eagle lax checking again !
As the error's wont allows you to extract the libs,   trying removing the TO92 from MAC97 TRIAC,
from the SCH file (with your editor) and see if the conversion completes,    I expect given all the above, that's why KiCad was spitting the dummy
Lachlan

PeteLawler, 2017-02-22:
> trying removing the TO92 from MAC97 TRIAC
Thanks for the tip, I'll get onto that shortly. Appreciate the other work you did to look into this. I figured diving in the deep end of a conversion would help both me learn about the procedure and maybe fix a bug or two along the way as well as identifying needed workarounds. I'm grateful you've taken my ignorance with good grace. Cheers!

lachlanA, 2017-02-22:
No problems,   be warned the Code is a mess!    I hack it all together from other scripts..
Thanks again for your input and help :)

Lachlan