lachlanA/eagle-to-kicad #27: Documentation of these scripts is really difficult to follow

JamesHagerman, 2018-01-31:
Would you be willing to accept a PR that cleans up some of the language in this repo's README.md? 

Specifically, Step 7 is as tedious as understanding a train wreck. The correct operation is closer to something like:

1. Open PcbNew without opening KiCad
2. `File > Open`
3. In the file type filter drop down, select `Eagle ver. 6x XML PCB files (*.brd)`
4. Select original, Eagle `.brd` file used for the rest of the ULP conversion process
5. Open the file

If no errors occur:

6. `File > Save As...`
7. Navigate to the `kicad/` directory the ULP scripts put it's output files (Note: If you changed the ULP output directory, look in there instead)
8. Select the `.kicad_pcb` file that's already there
9. Click `Save`

Continue to Step 8

lachlanA, 2018-02-03:
Yes,  No problems,  I'm always like improvements to my bad doc's !

Lachlan

JamesHagerman, 2018-02-06:
Okay, I'll sit down and do what I can!

lachlanA, 2018-02-06:
Thanks,    I need all the help I can get !!
Just ask my girl friend,  she has to put up with my spelling every day !!
LOL.. ;)

Lachlan.