lachlanA/eagle-to-kicad #30: kicad 4.7 does not exist anymore.

bob808, 2019-10-01:
ergo this software is obsolete. wish someone would make a way to convert eagle files to kicad.

mecworks, 2019-10-01:
@bob808, This bug should be open until the documentation is updated.  The readme still says "Only works correctly for version 4.7 of KiCad, will not work correctly if your using 5.xx version of KiCad to complete the conversion"

dandumit, 2019-10-02:
long time ago kicad was able to import directly eagle boards. just save your eagle board in new format and this will be imported in kicad pcb

bob808, 2019-10-02:
Yes, I had a super simple project and I had to repair some stuff to get it working.

lachlanA, 2019-10-31:
I will run a few checks, and update the doc's,  I thou most people were using the
eagle import in kicad, and using the library conversion script when they  had a large number of
eagle lib's to convert over to kicad, as I see there is no library conversion script in kicad, thou I may have mist it.
Lachlan

closed 2024-01-23 by bob808